odtu/EE463 #4: ?

MelisaSarici, 2017-11-11:
@ozank 
Hakan'ın rapora yazdığı bir commenti okurken düzeltmek ve sormak istediğim bir nokta doğdu hocam, power hesabını yaparken Voltajın total rms'ini currentin fundamental rmsi ile çarpıp integral ediyoruz ve kitapta da bu şekilde yazıyor, yani aktif power akımın fundamental componentindan gelmekte. Siz de derste bu şekilde söylediniz diye hatırlıyorum. Ancak bunun sebebi nedir kitapta yazmıyor. Bu nokta aklıma takıldı.

ilkersahin78, 2017-11-11:
Güzel soru, atlıyorum:

Mohadan ziyade aktif ve reaktif power ın tanımları için circuit kitabını tavsiye ederim, nilsson.

Sebep: Mesela yalnızca birinci ve üçüncü harmoniklerin olduğu bir sistemde: integral(sin(x).sin(3x).dx)=0.

Dolayısıyla güç harmonkten değil fundamentalden geliyor

MelisaSarici, 2017-11-11:
Teşekkür ederim hocam, şimdi anladım. Dediğiniz üzere işlemleri yapınca da çıkıyor. 👍

closed 2017-11-11 by MelisaSarici

ozank, 2017-11-13:
ilker'e :clap: :clap: