odtu/EE463 #5: team oluşturma

hakansrc, 2017-11-24:
@ozank 

hocam attığınız classroom linkinde accept this assignment dediğim zaman aşağıdaki gibi bir hata aldım. şuan da You are forbidden from performing this action on github.com hatası veriyor. grup oluşturamıyorum

![](https://lh3.googleusercontent.com/OpYh0s29CeEi9X-VbWnEonqgBhy_v-7Gzik5wJ7DAdaCag46uJ7r8zgk64t77WrdAHu3ofHIT_NYS8T7VWvtPzRhc4ZKFNJIQRupnHCzpm33xUAETpxH9WUEEpfq6_emsIgqmUnSTiUF5cAACQOFEkggmaf-OfKbAVbqPI1TILOXHnluFlOeTTloYlw2dkUfBSH82C77pQ0wh4-BWYRA7TqFsQ8LAewWKx84unUeAeDGa-Y37JaklduY-MBeoVWcVtdZO6i-z882cD92GWAQzC3mtdwxFfjSRCeqhUj5CHGCRR2sekoCIxTonkUOfGVcuhjSiOxayDxHwKoOBD29WAlGUe5ZOTM2ERBIO258PkaKCqXmCthKxjDFaHeS6pibZNRDNG5QfOQTOY6cbxEsK0_u1vhsPBYduapInB6YtA_BWB9hvudpP14wMpSxqp3NpJijEI6EfZyxAlBYxetOLlS4xCiNNcPc7ZAJjrUKCXZUEtu22cflP1PmRo5dP22T99w-cma609NYXPDCyZaLLj85WDThyAeoMHHGvd_sbCDF7CJpgY7nhqyi-eu2FeVBSNac4D8TJ9rZ3djBvDtRRf_RBXkkGzZbE5TC18tw2Q=w1187-h475-no)

MelisaSarici, 2017-11-24:
@ozank 
Ben de aynı mesajı alıyorum :/

ozank, 2017-11-24:
Herkeste aynı sorun olmuş, ben bi hata yaptım galiba.
Kontrol edip linki tekrar yollayacağım

hakansrc, 2017-11-25:
@ozank 

hocam "teğam 9" adında takım oluşturmayı denedim bi hata oluştu deyip tekrar denememi istedi.
tekrar aynı isimde oluşturmayı deneyince bu isimde bir takım zaten var diye uyarı aldım fakat bende görünmüyor o takım. sizde görünüyor mu acaba.

ozank, 2017-11-25:
Selam Hakan,
Şu anda takim: 1,4,5, 6, 8,11,12 görünüyor.
Sizinkini göremedim. Dediğim gibi gene aynı sorun yaşıyorsanız, kendiniz aranızda ortak bir repo oluşturup, beni de davet ederek öyle devam edebilirsiniz. İlla yolladigim linkten olmak zorunda degil.

Ozan

closed 2017-11-25 by hakansrc