odtu/EE463 #7: İlk soru

MelisaSarici, 2017-11-24:
@ozank 

Hocam ilk soruda, average current verilmiş. 
`V_{av }=2*sqrt(2)/pi*V_{s}*cos(\alpha) - 2*L_{s}*I_{d}*\omega/pi` denkleminde `I_{d}=40` koyarak alphayı çekmeyi düşündüm ancak average voltage gerekli mi yoksa ben yanlış yoldan mı düşünüyorum?

ilkersahin78, 2017-11-24:
a şıkkı için doğru

MelisaSarici, 2017-11-24:
@ilkersahin78 
Hocam average voltaj nedir o zaman istenen? L-R devresinden bir Z çekip Z*Iav ile bizim mi bulmamız lazım? Öyleyse eğer ben çok saçma bir değer buluyorum da 😀

ilkersahin78, 2017-11-24:
L nin averaj voltajı sıfır değil mi?

MelisaSarici, 2017-11-24:
@ilkersahin78 Evet hocam çok haklısınız, teşekkür ettim. 👍

ilkersahin78, 2017-11-24:
Rica ederim

ozank, 2017-11-25:
:+1: başkalarının da işine yararsa diye tekrar etkinleştiriyorum bunu

closed 2017-12-17 by ozank