odtu/EE463 #18: Project Repoları

ozank, 2019-01-04:
Arkadaşlar,
Projeleriniz için açtığınız repo adreslerini bugün bana bildirir misiniz? Donanımda çok ilerlemeden tasarımlarınız hakkında yorumda bulunmak istiyorum.

hhintoglu, 2019-01-04:
SPARK Industries
Huzeyfe Hintoglu, Sadık Akyar, Muhammed Hakan Karakaya
https://github.com/hhintoglu/EE463_Hardware_Project

ismail-ataseven, 2019-01-04:
BiB Power
İsmail Ataseven, Berkay Sağlam, Batuhan Bülbül
https://github.com/ismail-ataseven/BiB-Power

nailtosun, 2019-01-04:
Firing Angels
Nail Tosun
Ali Aydın 
Özgür Ertürk
https://github.com/nailtosun/EE-463-Hardware-Project

bulbulbahar, 2019-01-04:
The Mega Hurts
Bahar Bülbül
Etki Açılan
Hakkı Gülcü
https://github.com/bulbulbahar/EE463_HardwareProject

gurkandyilmaz, 2019-01-04:
K.A.R.P.U.Z.
Ali Aydın Yamandağ
Gürkan Durmuş Yılmaz
Sonay Ulukaya
https://github.com/gurkandyilmaz/EE463-Hardware_Project

anilcanbudak, 2019-01-04:
Freewheeling Co.
Anılcan Budak
Cem Akıncı
Murat Çolakoğlu
https://github.com/anilcanbudak/EE463-Hardware-Project

nazliogluahmet, 2019-01-04:
A.N.
Ahmet Nazlıoğlu
https://github.com/nazliogluahmet/EE_463_proje_AN

onurmetueee, 2019-01-04:
Smart Grid
Saliha İven Güzel
Ekin Su Saçın
Onur Külahlıoğlu
https://github.com/ivenguzel/EE463-Harware_Project

mertelmas, 2019-01-05:
METU LELS
Mehmet Elen
Mert Elmas
Yusuf Selim Karataş
https://github.com/yusufselimkaratas/463HardwareProject

nevzatsafasenyayla, 2019-01-13:
M.A.N.Power

Nevzat S. Şenyayla
Akın Şavklı
Metehan Kara

https://github.com/nevzatsafasenyayla/M.A.N.Power

closed 2019-10-08 by gkhnckl