odtu/EE463 #19: Solver Selection Changes the Results

edsenel, 2019-10-18:
In the simulations, the solver selection changes the results according to my simulations. Do we have to use any solver method or is it up to us? Because the results and values will be different from each other depending on the solver selection!

ghandeb, 2019-10-18:
ben şimdiye kadar hep auto (otomatik seçen) solver'ı kullandım.
bir de solver step size için variable step değil fixed step kullanman daha iyi olur. variable step olduğu durumda bazen step size gereksiz yere küçülüyor singularity hatası falan alabiliyorsun, veya steady state durumunda step size olması gerekenden büyük olduğu için accuracy sorunu yaşıyorsun. bu yüzden 1e-6 (5 veya 7 de olabilir, simülasyon süresine ve switching frekansına bağlı) gibi bir fixed step size sana istediğin accuracy'de bir sonuç verecektir

edsenel, 2019-10-18:
teşekkür ederim. auto solver seçip time stepi düşürünce sorun düzeldi

closed 2019-10-20 by gkhnckl