odtu/EE463 #22: Simscape Electrical Lisansı Hk.

nevzatsafasenyayla, 2019-11-25:
Hocam merhabalar,

Ben OSX işletim sistemli bilgisayar kullanıyorum. Simscape Electrical'ı ODTÜ'nün sunduğu lisans kapsamadığı için MATLAB'dan deneme sürümü almıştım. Fakat şimdi 3. proje üzerinde çalışmak için MATLAB'ı açtığımda deneme sürecinin bittiği uyarısı veriyor ve dolayısıyla herhangi bir şey yapamıyorum. Simscape Electrical'ı satın alma konusunda ise 1400$ gibi bir ücret istiyor MathWorks.

Bu konuda sizin bilginiz olan, okulun sunduğu başka herhangi bir çözüm var mıdır?

Şimdiden değerli zamanınız için çok teşekkürler..

Selamlar,
Nevzat.

ghandeb, 2019-11-25:
merhaba nevzat, oncelikle lisans satin almak gibi bir seyi bence aklindan bile gecirme, inanilmaz pahali. bir de aklima takilan baska bir soru: simscape electrical demissin ama ee463 kapsaminda genelde specialized technology toolboxini kullaniyorduk biz, hangisini kastediyorsun? macos icin nasil bilmiyorum ama okulun windows tabanli matlab lisansi simscape electricali kapsiyordu ama specialized technologyyi kapsamiyordu. onun icin malum ortamlardan farkli bir matlab lisansi bulmak gerekiyordu. macos icin de muhtemelen farkli degildir, okul lisansiyla cozebilecegini sanmıyorum. kesin bilen biri fikir verse daha da guzel olur tabi

nevzatsafasenyayla, 2019-11-26:
Hande Hocam merhabalar, Windows'ta nasıl bilmiyorum ama MacOS'ta 'Simscape Electrical' ayrı bir Add-On olarak var ve Specialized Power Systems Toolbox adı altında kullanıma sunuluyor. Odtü'nün sunduğu lisans MacOS'ta Simscape içeriyor ama Simscape Electrical'ı içermediği için bizim projeyle alakalı çalışma yapılamıyor ne yazık ki. Şimdilik ben Simscape Electrical Add-On'unu silip tekrar yükledim ve free-trial verdi ama sanırım dönem bitmeden yine aynı sorunu yaşayacağım, kalıcı bir çözüm bulmak gerek o nedenle. İlginiz için çok çok teşekkür ediyorum.

ghandeb, 2019-11-26:
aklıma gelen çözümler genelde illegal :) windows için bulunuyorsa macos için de bulunur muhtemelen ya. bir de böyle durumlarda virtual machine üzerinden windows çalıştırmak vs gibi şeyler yapanlar olduğunu hatırlıyorum, hiçbir çözüm bulamazsan belki onu araştırabilirsin. pek yardımcı olamadım maalesef mac olunca, windows olsa gel benden al derdim

closed 2019-12-11 by gkhnckl