odtu/EE463 #25: THD Simulink and Analytical Comparision HW1

ceyhunkocc, 2020-10-29:
Hello  everyone,
I have a problem about  THD calculation. In  part c, I have done  the  calculation directly from voltage because  we  have a resistive load, it is just current's multiplication. You can  find my calculation below, however simulink gives  nearly  0.43. What may  cause that problem?
CUMHURİYET BAYRAMIMIZ KUTLU OLSUN!
[CamScanner 10-29-2020 11.44.02.pdf](https://github.com/odtu/ee463/files/5457091/CamScanner.10-29-2020.11.44.02.pdf)

ahsabirli, 2020-10-29:
Hocam ben de sizin gibi aynı sonucu bulup Ned Mohan'nın kitabına yöneldim, sayfa 40 sondan ikinci paragrafta belirttiği gibi THD hesaplamasında sizin de kullandığınız THD=sqrt(Irms^2 - I1,rms^2)/I1,rms formulünü kullanmak için DC değerinin 0 olması gerekiyor. Bu yüzden bulduğunuz voltage veya current waveform un RMS değerinden DC değerini çıkartıp işleme öyle devam ederseniz 0.43 e ulaşıyorsunuz.

Sizin de bayramınız kutlu olsun.

ceyhunkocc, 2020-10-29:
Çok teşekkür ederim.

closed 2020-11-04 by ceyhunkocc