odtu/EE463 #26: Why do we use discrete solver?

mehmetkl, 2020-11-07:


mehmetkl, 2020-11-07:
I am okey with using a discrete solver but I wonder the reason. Why we prefer to use a discrete solver, instead of a continuous one?

furkankarakaya, 2020-11-07:
Hi Mehmet. There are actually several reasons for using a discrete solver instead of the continuous one. First of all, discrete solver time helps us to foresee the simulation duration before simulating. Let's say we are running a simulation for a 10us duration with 1ns sample time. The simulation will be performed for 10,000 points. Based on the results we obtained, we can either increase the sample time or decrease it. Sometimes, if not limited, a continuous solver tries to solve a circuit with a too-small step time and the simulation takes much longer than we expect. A solution for this problem might be defining a minimum time step for the solver. 

Moreover, the definite sample time helps us to understand and investigate some results in the circuit. Having known the duration of a sample we can actually calculate the derivatives even by hand, or we can export the data to the workspace and do some integration. For a continuous solver, integration can only be done by having the time data as well. Moreover, sometimes we observe the effect of sampling time in our harmonic analysis. Instead of observing something we don't expect, we prefer to use a predefined sample time.

closed 2020-11-07 by mehmetkl