odtu/EE463 #28: Input voltage regulation for Term Project

alpercak, 2020-12-13:
Hello everyone,

I am trying to regulate the varying input voltage by using simple diode rectifier. After rectification, a buck converter is implemented to have 24V output voltage. So far no problem has been seen. 
However, the voltage after rectification still have variance e.g 150V to 330V. Should we regulate this voltage to a certain value? 
I think its not possible to have a certain DC voltage with using diode rectifier.
Do you have any suggestion like isolating the circuit with transformer or so ever?

Thank you

ceyhunkocc, 2020-12-13:
Hello,
As far as I understand, there are different battery charging methods which are constant current, constant voltage, constant current-voltage etc. In this project we want to charge our battery with constant current method, so you need to vary the duty cycle to keep battery current constant. In this situation, the voltage drop on the battery, at the output of the buck converter, may be very lower than 24V.

cayunus, 2020-12-13:
> I think its not possible to have a certain DC voltage with using diode rectifier.

Is not this the explanation of why the control of buck converter is needed? I think the adjustments of duty cycle should be done according to output voltage of rectier so that the current supplied to battery is constant. If the problem is voltage ripple, DC link capacitor decreases ripple.

alpercak, 2020-12-13:
Thank you for your answers. You are both right I suppose. I arrange duty cycle using a basic controller circuitry. So far, i can get 22-23V output voltage with 1.8-1.9A current on the load. Now I have two questions;

1) I haven't connected the battery yet. I am using 12 ohm resistor for the load. Should our design include a load like resistor? I am asking this question, because, when I dont connect load, the output current which is now capacitor current and voltage goes infinity. 

2) How should we select DC link capacitor? Do we have a method to calculate it? I now use 2200uF DC link capacitor just thinking its enough and easy to find. It seems there is not much ripple on the voltage just before it reaches buck converter.

ceyhunkocc, 2020-12-13:
To connect DC-link capacitor, there are calculation methods with desired current and voltage ripple at the output of Buck Converter which can be found here:
https://www.ti.com/lit/an/slva477b/slva477b.pdf?ts=1607698500995&ref_url=https%253A%252F%252Fwww.google.com%252F
For other question, I did not connect anything between DC-link capacitor and battery, however in real design, you should add a sense (shunt) resistor between battery and ground to take current feedback.

alpercak, 2020-12-13:
Thank you Ceyhun, this document is great! 

Actually i ment the output side of the buck converter.  Normally in the buck circuit, there is a low pass filter and a load resistor after switching element. But for the project we only have a battery on the load side. I confused that should we keep the load resistor in our design? or the output elements of the buck converter should only include an inductor and a capacitor?

ceyhunkocc, 2020-12-13:
From my point of view, in this project we are using battery as load, which means the inductor and capacitor is enough.

alpercak, 2020-12-14:
I figured it out now. I disconnected the output resistor and the duty cycle arranged by referencing the output current.
Thanks again.

closed 2020-12-23 by alpercak