odtu/EE463 #29: Controller Selection for Buck Converter / input voltage issue (Term Project)

alpercak, 2020-12-24:
Hello everyone,

I have struggled with the selection of the controller. So far I found integrated controllers like LM5117 accepting maximum 100V as input voltage. I tried to search isolated gate drivers, however, I couldn't get any progress so far. 

Does anyone help me to guide for the selection of controller IC ? 

Thank you

tokgozfurkan, 2020-12-24:
Hi,
Although it would be better to use controllers specified for Buck converter, you are not limited to these types of controllers. For example, [TL494](https://www.ti.com/lit/ds/symlink/tl494.pdf?ts=1608778620154&ref_url=https%253A%252F%252Fwww.google.com%252F) is a simple PWM control IC with two comparators and an oscillator. Try to widen your search range depending upon the application.

ahsabirli, 2020-12-24:
If you want to use voltage regulator IC (which consists MOSFETs and filters aswell) instead of a controller IC, you can consider back to back step down converters. First one with open loop control i.e. fixed duty cycle to reduce de voltage to 30-50V and second one with integrated controller and filter. But after first converter you should also use a moderate filter. For first contoller you can use simply a MOSFET, oscillator and mos driver. Second converter can be like LMZ14203H. which supports input voltage up to 42V and can supply 3A with adjustable output voltage. Efficiency of LMZ14203H can go up to 97% for 24V output and has very good output voltage ripple values like 80mV p2p. I dont know if this topology is efficient or cost effective but it will be interesting to inspect :)

closed 2021-01-09 by tokgozfurkan